DoESLiverpool/Tosca #5: Build limit switches for the spools

amcewen, 2016-02-13:
Machinekit expects to have limit switches so it can home properly.  We need to work out a way to find out when the spool has hit the end.

amcewen, 2016-02-20:
https://github.com/machinekit/machinekit-docs/blob/master/src/config/ini_homing.asciidoc is the Machinekit documentation for how homing works.  It looks like it will always home one axis at a time - you can set the order that it homes the axes or define that you don't care for certain sets of axes, but I couldn't see a way to get them to home at the same time.

goatchurchprime, 2016-03-01:
We can't pull the gondola off the playing area and all the way up to the pulleys, so the only way this will work is by having some markers attached to the strings that are at a known location which can be detected when in range (eg a little magnetic grain and a hall sensor) and we put them at a place where the get triggered when the gondola is at the top of the board, but not pulled off.  As long as the system always turns on with the gondola below these positions, then the homing cycle will work and not run off the edge.

DefProc, 2016-03-01:
Would a microswitch at the bottom of the gondola work?

Assuming a level base, it would let you drop both motors till the switch
operates, then pull both a bit up and lower one till the switch operates to
know overall position.

amcewen, 2016-03-02:
I think the big stumbling block (unless I'm misreading the Machinekit documentation) is how you get it to run a homing cycle that's anything more advanced than "run an axis until it triggers a limit switch"

goatchurchprime, 2016-03-02:
It's basically all about getting it to do a little dance on the motors and switches that ends with it being in a known position.  

I reckon a more advanced homing cycle is possible to code through the Python interface where it runs combination movements on the axes (eg pull both in or both out), and spots trips on the limit switches.  Then it can click the "now it is homed" feature [c.home(0); c.home(1)] which just declares it done.  

http://linuxcnc.org/docs/2.6/html/common/python-interface.html