DoESLiverpool/Tosca #8: Will SVG be OK as the graphics format for front end software?

helicalbytes, 2016-02-15:
I've been looking at formats to represent graphics (including text) that the polargraph can draw. Just want to check that it seems OK to others too.

The best overall compromise seems to be SVG. This can represent lines, curves (quadratic and cubic Bezier) and arcs. The Bezier curve segments can be approximated by a series of arcs, so that should work quite well with g-code. Open source fonts can be easily converted to SVG too - this results in well-formed XML that is easily parsed, so getting an individual letter's shape is easy. And for non-text output, most graphics packages can output SVG, which should hopefully be manageable in a similar way, so it should be possible to create graphics using multiple software packages.

amcewen, 2016-02-20:
That seems to make sense.  SVG would be a good format for any web display of it too.  And you can always use Inkscape to conver the SVG into gcode (assuming there isn't an easier option of just reading in the probably-fairly-simple SVG and churning out the Gcode ourselves :-)

helicalbytes, 2016-02-20:
OK. I have an open source simulator installed now that can read g-code files, so hopefully that will allow checking any SVG to g-code conversions that we have. I'm working mainly on converting fonts (which have been converted to SVG format) to g-code. The SVG is very simple - a single glyph is essentially just an SVG path element made up of line and Bezier segments. I've got basic SVG generation done now and am just starting to work on the conversion to g-code.
[image]

closed 2016-02-23 by helicalbytes