DoESLiverpool/Tosca #10: Gondola for chalk spray?

jackie1050, 2016-02-16:
Would be good for outside of buildings

huffeec, 2016-02-17:
I like this. Have been thinking about how we can make a print head that moves quickly. If we went for a chalk/paintjet head it wouldn't have to be in contact with a potentially bumpy wall.

jackie1050, 2016-02-18:
Indoors could use black paper and uv light

jamieisboss, 2016-02-27:
I have some chalk pens, I'll try to remember to bring them if you want to try them out. They kind of drip feed so wouldn't work at 90 degrees to the glass for long, but 45 degrees should be alright. A couple of them fluoresce under uv so if that's what you're going for they could be a good choice? You won't need a black background if you have a good uv source.,you do have to be careful about reflection of the uv source when the glass is bare. 

Stage e40 bulbs work a treat but I'm pretty sure your eyes melt if you look at them when they're on. I can ask a proper lighting guy if it'll help, all qualified and whatnot 

For grip, maybe an air pump to create suction between the gondola and glass when drawing? Relay it on and off for speed or precision, mitigate the noise by putting it somewhere else and using a reasonably long tube?

jackie1050, 2016-02-27:
I've bought some uv pens already - going to try them out today, but pro

jackie1050, 2016-02-27:
I've bought some chalk pens now, but probably worth testing yours too. Interesting to know you don't need a black background - don't want to do anything dangerous though.

amcewen, 2017-06-12:
I have a couple of half-full spray chalk cans, [courtesy of @andatwhatcost](https://twitter.com/andatwhatcost/status/873217858360291328)... :-)