DoESLiverpool/Tosca #11: Capacitive touch through a window to control it

amcewen, 2016-02-18:
For one put inside a window, it'd be interesting to make it more interactive - maybe use capacitive touch pads on the inside of the window to let people control it, without needing anything on the outside of the glass...

amcewen, 2016-03-09:
Now that we're running Knut as a poetry-battle, we won't need control-through-a-window.  Still, this would be a great thing to do sometime in the future (so moving milestone accordingly :-)

jamieisboss, 2016-03-18:
Was going to suggest for the glass, not sure how a poetry battle works, but could be behind the board so long as you remember where it is. 

Cooler; under a table, if the tabletop wasnt too thick you wouldnt have to touch it just practise and it'd look very cool magically controlling the gondola just waving your hand in the air.