DoESLiverpool/Tosca #13: Design and build two sets of "corner pulleys"

amcewen, 2016-02-20:
The motors and spools sit on the floor and the thread then loops over some "pulleys" (they'll likely just be eyebolts) in the top corners of the drawing area.

@DefProc used eyebolts on a laser-cut wedge for the TLRN to good effect, so we should build something similar to hold the two threads for the polargraphs.

amcewen, 2016-02-20:
A pack of 10 M6 eyebolts http://www.ebay.co.uk/itm/160870170069 has been ordered, should be arriving on Tues 23rd.

DefProc, 2016-02-21:
Ah. I have some spare

goatchurchprime, 2016-02-22:
Might need these to be real pulleys that introduce less friction into the system. There is a huge difference in force pulling in and letting out, and it's the letting out that's critical to the speed.  The string needs to run easily when the gondola is put into falling mode.

amcewen, 2016-02-24:
The eyelets have arrived and are in the Tosca storage box.

DefProc, 2016-02-24:
The braided fishing line is UHDPE, so it should provide it's own bearing surface, so it's unlikely to significantly reduce the amount of friction in the system (although there is some wear, so for ultimate longevity, pulleys are *technically* better). 

What's the requirements for mounting the top brackets? Self-adhesive pads onto a painted/glass surface? Screws and rawl-plugs into a drilled hole in blockwork?

How many strings do you need to control? The current implementation seems to have two per motor. Is that just because the motors are mounted to double pulleys? Will you switch to a single line? *(The 0.45mm line **is** strong enough)*

amcewen, 2016-02-27:
The two strings help with the stability of the gondola, so we're sticking with two string option.

In discussions today we reckoned that the current TLRV versions are working well, and we won't know any more about how they need to be attached until we've had the site visit (issue #19), but then might just need to get a copy of the files and cut a second set.

DefProc, 2016-02-27:
Ok, they are designed for a single centre screw mount, but could be
adhesively mounted is the base surface is good enough

amcewen, 2016-03-13:
The Knut polargraph will be mounted on the window, so we should be able to reuse the existing mounts with some command strip (I've bought some, it's currently on my desk though I'll stick it in the Tosca box in the workshop)

For the main Tosca polargraph, @epsaul, presumably we can just add some eyelets to the top corners of the drawing board?

amcewen, 2016-03-15:
We're going to use the eyelets chased into the drawing board, so this is now sorted!

closed 2016-03-15 by amcewen