DoESLiverpool/Tosca #14: G-code unit scaling

helicalbytes, 2016-02-22:
I have some text from SVG fonts ready to  convert to g-code for testing. Just want to check the scaling. That is, how many g-code units equal one physical unit (e.g. cm) on the drawing surface?

goatchurchprime, 2016-02-22:
The scale is set in mm.  One G-code unit for 1mm.  

On the current whiteboard setup we can safely go X-200, to X200 and Y-200 to Y200

Z is ignored, but takes time, so go up and down in it by 1mm.  May be used for the pen servo later.

See Issue#15 about corner rounding and possibility of non-outline letters.  (More like hand-writing type letters).  

For emphasis we'd do more interesting shaded letters where areas are hatched or painted in.

helicalbytes, 2016-02-23:
Thanks for info. Regarding single line (as distinct from outline) fonts, I did have a look for some of these a few days ago. I found a few that are used for engraving and other similar applications, but I wasn't sure if there are any that are widely known and used?

closed 2016-02-24 by helicalbytes