DoESLiverpool/Tosca #16: Design a gondola with 4 wires and a weight

goatchurchprime, 2016-02-22:
I have bodged this with gaffer tape and a pair of metal pliers, and strung up the extra strings to keep the pen level.  There is a lot less jerkiness.   This has the potential to make something more rigid and not dependent on the friction of the pen tip.  

There's still a free movement of rotation that allows oscillation that I'd like to get rid of -- even though it might be less important when the rotation is about the axis of the pen.  It still would make it difficult to run with more complex structures with 2 pens in them.  

Maybe the CofG needs to be lowered down an hung in such a way that it is lifted by the strings if it is anything other than level.

goatchurchprime, 2016-03-01:
We need to get the hanging eyelets closer together on the line perpendicular to the pairs of strings.  At the moment the pen angle to the board changes quite a bit from the bottom to the top.  

Needs hooks on the front strings of the gondola to allow for fine adjustment of string length to get the pen level.  

Don't forget a place for the Jeenode and 4xAA cell battery pack.  This has run for at least 24 hours without stopping so there's no problem with the power supply.

amcewen, 2016-03-01:
I think it'd be useful for the pen to be mounted angled slightly downwards rather than perfectly horizontal - that'll help ink flow to the tip.