DoESLiverpool/Tosca #17: G-code from text -- for testing

helicalbytes, 2016-02-23:
Attached are some g-code files rendered from SVG glyphs for testing.

I'm not familiar with the g-code format so what I've done I'm sure isn't best practice! However the output does seem to render OK in Camotics, which is an open source CAM simulator (screenshots below). Please let me know about any mistakes I've made.

In particular, for 'move' operations I've used the following 'pattern':

G0 X-121.40 Y-36.84 Z1               // z = 1, fast move (G0) to start of new path section
G1 X-121.40 Y-36.84 Z0 F100     // z = 0, with feed = 100
G1 X-124.91 Y-36.67 F100           // move to end of first line

There are three files at different sizes in the zip, each showing the word 'tosca'. The origin for these is approximately in the center of the text.

tosca-300.gcode.txt : total width approx. 300mm (g-code units)
tosca-150.gcode.txt : approx. 150mm wide
tosca-75.gcode.txt: approx. 75mm wide

[image]
[tosca.gcode.zip](https://github.com/DoESLiverpool/Tosca/files/142614/tosca.gcode.zip)

amcewen, 2016-02-26:
http://www.cambam.co.uk/forum/index.php?topic=1557.0 has a selection of single-stroke fonts, and if you install the latest version of Inkscape then you'll also get the Hershey text extension (in the Eggbot extensions), but that seems not to be a simple font file - it's at https://github.com/evil-mad/EggBot/blob/master/inkscape_driver/hersheydata.py (which has links to the source it's derived from in the comments) in case that's of use

helicalbytes, 2016-02-26:
Thanks for those. Have put a couple of cambam ones through the font software I use to convert them to SVG and that works fine. They render OK using my code too. Will put some g-code generated from these over next couple of days.

amcewen, 2016-03-15:
Now we've got the SVGizer for text, I think any remaining text work will be covered under issue #23

closed 2016-03-15 by amcewen