DoESLiverpool/Tosca #18: Pen up pen down system from Z-axis control

goatchurchprime, 2016-02-24:
In original design this is done using a servo that operates a lever to push itself (and the pen) out of contact with the surface.

Using the z-axis to control the pen position allows for pressure changes and quick response during travel (eg to make dotted lines).  This is going to be more dynamic than using tool changes or coolant on/off settings as a proxy.  

The home value would be where the pen is only just in contact with the surface.  There should be travel on either side so that we can exert variable forces.

It's probably going to work wirelessly using a Jeenode pairing, with one Jeenode on the gondola receiving a Z-value and controlling a mini servo motor accordingly.  The transmitting Jeenode would read from the BeagleBone.  We can just use it to read two pins (step and direction), which can be decoded in an arduino with step connected to one of the IRQ pins that increments or decrements a register up through a distance of 1mm (in machinekit world).  

This BB INI file can have all the settings for number of steps in a mm and the speed of acceleration (high as there's no intertia).  The arduino initially would use a cable to the gondola to control the servo directly.  

We therefore should begin by wiring up an arduino and make a sketch to receive step and direction (pins P9.26, P9.24 according to the HAL file) to control a servo on the table right now.

DefProc, 2016-02-24:
I have some jeenodes, and lipo batterys that would be suitable.

goatchurchprime, 2016-02-27:
This is now working with https://github.com/DoESLiverpool/Tosca/blob/master/jeenodezdecoder/jeenodezdecoder.ino and https://github.com/DoESLiverpool/Tosca/blob/master/jeenodepenactuator/jeenodepenactuator.ino which have been gaffertaped to the gondola with a 5V 4xAA battery pack (note 7.2 volts melts the servo).  

There's a home button on the zdecoder that transmits a home signal to the gondola.  It's not perfect, and there's some configuration relating z values to pen positions.  The RF code needs to send ACKs to make it more reliable.

closed 2016-02-27 by goatchurchprime