DoESLiverpool/Tosca #19: Arrange a site visit

amcewen, 2016-02-25:
We need to work out how we can rig it, etc. so visiting the venue would be really handy :-)

JennyHarris, 2016-02-26:
Shall I put you in contact with Richard the venue manager?

amcewen, 2016-02-26:
Ooh, if you could, that would be great!

amcewen, 2016-03-01:
Site visit arranged for 1pm on Weds 2nd March.

amcewen, 2016-03-03:
Visit done!

There are some photos I took at https://www.flickr.com/photos/amcewen/albums/72157664813872800

The window version shouldn't be too tricky: we can easily run power to one of the window ledges, and put everything on there.  The windows are over 10ft wide and more than 7ft high, so there's a nice big area to draw on.

For the main performance area - the screen can't easily be moved (and Lizzie has some photos she'd like to project onto it anyway).  We think the best approach is going to be to hang the drawing area to the left of the screen (when looking at it from the audience, like in https://www.flickr.com/photos/amcewen/24811058904/in/album-72157664813872800/).  There are hefty scaffold poles mounted vertically (as can be seen in https://www.flickr.com/photos/amcewen/25415458676/in/album-72157664813872800/ although those are to the right of the screen) which are 48mm diameter.  The ones to the left are 10 feet apart, so we could fairly easily hang a full size sheet of plywood between them.  We need to avoid the fire exit, but that's unlikely to be a problem.  The venue have the right kit to get access to them to do the hanging.  It's probably best to build it as a self-contained unit just needing power, as it'll be trickier (although not impossible) to have things on the floor below it (particularly on the left side, as that'll be off the edge of the stage and near the fire exit)

There's WiFi available (and should be decent, as they use it to livestream things).

They also have flexibility with how it could be lit, although we weren't sure what we want beyond maybe having a spotlight pointed at it :-)

amcewen, 2016-03-03:
Issues #3, #7, #13, and #33 updated with details learnt after the visit, so I think we can close this issue now.

closed 2016-03-03 by amcewen