DoESLiverpool/Tosca #20: Can we draw a map with it

amcewen, 2016-02-25:
Need to get the openstreetmap data and the area of Liverpool that we want to draw, and process it into SVG and gcode.

goatchurchprime, 2016-03-01:
Please submit an SVG of the map here

amcewen, 2016-03-01:
I've created an Examples directory, and stuck an example SVG of the centre of Liverpool into it - https://github.com/DoESLiverpool/Tosca/blob/master/Examples/Liverpool_city_centre.svg

amcewen, 2016-03-01:
From Lizzie:

"the area for the map will need to be between Lodge Lane in L8 and Upper Duke Street. We only need to show Upper Duke Street as far as the Anglican Cathedral but it'd be great to get the whole of Lodge Lane in. It's also important that Hope Street's on there."

So that's what we'll need to generate for the performance.

amcewen, 2016-03-21:
Can we draw a map with it?  We can.  And we did!

closed 2016-03-21 by amcewen