DoESLiverpool/Tosca #22: Get some pens that can draw on windows

amcewen, 2016-02-26:
The stepper-based version is going to draw on the inside of a window (so passers-by can see it - and maybe even interact with it).  So we need to get some pens that can draw on the window (and that can be cleaned off afterwards too :-)

jackie1050, 2016-02-26:
'll try to get some scraps of black foamcore or correx to try them out on too

amcewen, 2016-02-26:
There is a spare glass bed for the RepRap that I'm sure we could try, and I guess there'll be plenty of glass bottles in the recycling for initial "does this write on glass, and can we clean it off?" testing :-)

Plus some windows in Dinky and the main office that we can try rigging it up onto...

jackie1050, 2016-02-27:
I have bought some pens. Currently at DoES so hopefully will be able to try them out later.

amcewen, 2016-02-27:
Cool.  The RepRap glass is in the 3d printing supplies cupboard under the printers, if you want that to try out.

I've got some stuff to do, but should be in DoES around 3ish

jackie1050, 2016-02-27:
Chalk markers do write on glass

closed 2016-02-27 by jackie1050