DoESLiverpool/Tosca #23: Decide what the window polargraph is going to draw

amcewen, 2016-02-27:
The stepper polargraph is (hopefully) going to be sat in one of the windows of the Music Room, so passers-by can see it.

It's probably going to draw poems, but we need to work out how that works.

There's been discussion so far about maybe writing some Markov-chain software, or something similar, that auto-generates text given some source examples.

That fits nicely with Adrian Henri's work like [the New, Fast, Automatic Daffodils](http://www.adrianhenri.com/writer-poems-automaticdaffodils.html) where he mashed up Wordsworth and car adverts.

If we could find a suitable source text for advertising copy, and similar for Adrian Henri and Wordsworth, then we could autogenerate modern equivalents...

amcewen, 2016-02-27:
https://blog.arduino.cc/2016/02/17/macchina-poetica/ and http://www.bbc.co.uk/rd/blog/2016/02/irfs-weeknotes-number-213 might also give some ideas about auto-generated text.  And I built (well, did some simple copying ;-) a "Horse Tweets" twitter markov bot called [@policyomatic](https://twitter.com/policyomatic).  That takes a load of manufacturing policy documents as its source text, and then spits out auto-generated samples.  The original twitter-bot script that I stole is https://gist.github.com/koffeinfrei/a0ebc5c9295c067fe5ff

epsaul, 2016-03-08:
Link to the poetry algorithm http://www.aaai.org/ocs/index.php/HCOMP/HCOMP14/paper/viewFile/9049/8986

helicalbytes, 2016-03-15:
[hello-tosca-0.25.gcode.txt](https://github.com/DoESLiverpool/Tosca/files/174593/hello-tosca-0.25.gcode.txt)

helicalbytes, 2016-03-15:
[hello-tosca-0.05-gcode.txt](https://github.com/DoESLiverpool/Tosca/files/174626/hello-tosca-0.05-gcode.txt)

jackie1050, 2016-04-30:
Poems created and performed at the event - joint first prize won

closed 2016-04-30 by jackie1050