DoESLiverpool/Tosca #24: Recode the Jeenode sending to use ACK

goatchurchprime, 2016-02-27:
Continuation of issue 18:  https://github.com/DoESLiverpool/Tosca/issues/18

Current code in https://github.com/DoESLiverpool/Tosca/blob/master/jeenodezdecoder/jeenodezdecoder.ino uses:

        rf12_sendStart(1, payload, payloadCount*sizeof(int));
        rf12_sendWait(RF12_NORMAL_SENDWAIT); // wait for send completion

Some packets get missed which means the pen doesn't move in time (and gets set at the next heartbeat).  

A solution would be to require ACKs to make sure the packets are getting through (and so it can retry multiple times if it doesn't).  This will improve the relability.  
    http://jeelabs.org/2011/01/12/relaying-rf12-packets/

amcewen, 2016-03-03:
I've marked issue #31 as a duplicate of this, and edited this one to show the problem that needs solving, rather than one of the possible solutions (now that we'd ended up with two issues trying to solve the same thing).

amcewen, 2016-03-03:
Personally I don't think that switching to IR over RF would be a good idea.  With RF we at least have *some* experience, whereas IR is switching to a whole new domain for all of us.  And when did anyone last have a phone with an IR port?  Or use the one that it had?  I think there's a reason that Bluetooth/Wifi/etc. won out over IrDA :-D

I've got a pair of XBee modules which might be more reliable than the RFM12 modules.  They're on @goatchurchprime's desk at present.  They're targetted at higher bandwidth than the RFM12 modules, if nothing else...

amcewen, 2016-03-15:
Moving to XBee seems to have improved matters considerably.

closed 2016-03-15 by amcewen