DoESLiverpool/Tosca #27: Sort out advanced OS matters on the BB

goatchurchprime, 2016-03-01:
1.  Need a way to obtain the IP number of the BB without needing to search for it on NMAP.  Adrian mentions some sort of MDNS technology.

2. Need a way to disconnect from the BB while leaving the terminal window and it's X-windows "axis" UI interface running, and then be able to reconnect to it and carry on.  Otherwise when running a long operation we're stuck with having to have a separate laptop connected at all times just to see the UI and do nothing else.

amcewen, 2016-03-01:
For (1), I've used "apt-get install libnss-mdns" on the Raspberry Pi.  That means you can then connect to it as hostname.local, so it'd be machinekit.local.  If that works then we should look at giving each polargraph its own hostname (tosca-stepper and tosca-servo?) so we can connect to them both (once we've got more than one...)

amcewen, 2016-03-01:
And for (2), I've not used it, but a quick google suggests that https://en.wikipedia.org/wiki/Xpra would do the trick

amcewen, 2016-03-06:
The mDNS stuff is already installed on the Beaglebone, so it was available as beaglebone.local.

I've updated the hostname (following the steps in https://sheldondwill.wordpress.com/2013/12/14/43/ - which is basically "edit /etc/hostname and /etc/hosts, then reboot") and set its hostname to tosca.

So now to ssh in you just need to do ```ssh machinekit@tosca.local```

We'll need to change the name on the other Beaglebone (to knut, for the stepper-motor version) once we've got two Beaglebones on the network :-)