DoESLiverpool/Tosca #29: Buy the parts we need

amcewen, 2016-03-01:
There'll be a chunk of stuff that we need to buy for the two polargraphs.  As the performance isn't too far away, we need to get things ordered sooner rather than later.

Add things to this list (edit the issue) and we can get them ordered, and then check them off once they've arrived.  Links to where to get them from preferred (and saves @DoESsean asking you where to get them from :-)

Need ordering:
 * [ ] Beaglebone green x 2

Ordered/arrived (checked off when they arrive):
 * [ ] Eyelets
 * [ ] Chalk markers

amcewen, 2016-03-01:
@huffeec it'd be good to know what stuff we need for the servo motors pulleys so we can get that ordered.  @DefProc can no doubt advise on that too, given he bought stuff (and has design files, so bug him for them for issue #3 ;-) for the stepper pulleys, etc.

amcewen, 2016-03-01:
@goatchurchprime Where did you get the Beaglebone Greens from?  Is 2 going to be a good number to buy?

goatchurchprime, 2016-03-01:
Beaglebone greens got from here:  http://www.mouser.co.uk/Search/ProductDetail.aspx?R=102010027virtualkey67130000virtualkey713-102010027  £64 for 2 with vat.

2 puts it over the £50 limit for free shipping.  I think we got enough floating around that we don't need more spares.  

If you're ordering from there, I'd like one of these big yellow OLED displays:
    http://www.mouser.co.uk/Search/ProductDetail.aspx?qs=8UoQy1KDZL7fwY%2fpFrKBgw%3d%3d

amcewen, 2016-03-03:
@DefProc Where did you get the command strip from that you used to mount the corner pulleys onto windows?

amcewen, 2016-03-09:
I've ordered two rolls of paper (wasn't sure if the wider one would bleed through at all) for us to use as a writing surface for the main performance and/or for use to draw things on afterwards when it's installed in DoES somewhere

amcewen, 2016-03-13:
@epsaul Do we know how we're going to hang the drawing board from the scaffolding poles?  Do we need to buy any clamps?  Cable?

amcewen, 2016-03-14:
Servo, new stepper driver board and XBee-based Arduino boards ordered.

jamieisboss, 2016-03-15:
I can reccomend http://www.scaffolding-direct.co.uk/ , you likely dont need actual scaffold pole (no qualified to say hat but you could always add support), aluminium would be much easier because the steel is just so heavy, they wil cut to size for you, have a whole range if clamps and fittings and the delivery is usually a lot but theyre based in bromborough and will allow collection if you call them.

Unfortunately can't make any promises, it'd be for a society thing and need to secure funds -  but i'll probably have a use for any pole and clamps you buy, and would be happy to buy them from you if you don't need them after this and they're not being altered dramatically and I get the funding together. Three caveats, but t's actually quite likely

JennyHarris, 2016-03-15:
How heavy is it?  Good quality ratchet straps may do the trick.  I've used these to hang big mdf panels from scaffolding before.

amcewen, 2016-03-15:
I don't think it'll be too heavy.  Discussing with @epsaul just now, he's going to source the wood and some thick enough cord/rope to use to fasten it up.

I've ordered the XBee and Beaglebone Green boards too.

jamieisboss, 2016-03-16:
the ply will be 40KG +- 15% for the type of wood and treatment google says 42KG for 2 sheets,  if you get 12mm (vive la revolution!) if you went for 3/4" (18mm) that'd make it 66kg for ply 

Then if you're planning  to screw hat to some 3"x2" thats 1.4kg/m (im just googling), assuming you'd do a garden gate back with 20mm clearance so you coudnt see it from he front and i's all 2m for ease  and you ended up with PAR redwood, thats a 10Kg. 

Adding in the scaffold if you use aluminium would be 2kg/m  (needed wolfram for that) you'd need about 16m if you add in the props, joining clamps and a base with wheels , getting up to 85Kg 

If you use steel scaffold that's 4mm thick its 4.37Kg/m so 70kg for the scaffold, call it a round 120Kg for the lot without any functionality besides wheeling.

Just as a caution that if you've got to get that on and off the stage, and it'd be 3m tall by that point so wont go through a doorway, you probably want to think about the weight and moving it , plus if its at the phil you'll kill the stage finish unless you get the right castors

jamieisboss, 2016-03-16:
I realise none of that was helpful, so maybe if you made it so the middle folded, and the wheel base that was on one side could relocate to the other top corner and the bottom to its 90 degree so then you could strap it/clamp it at the bottom and wheel it out?

jamieisboss, 2016-03-16:
so 3 corners would be like this fitting

http://images.scaffolding-direct.co.uk/product-images/1000-1000/328-176d.jpg

then you have  short piece to join to a regular T whih has corners and castors, then they can come out of the bottom, go on the side and fold it in the midddle. 

That makes more sense than the first way i described it. Unless he stage exit is big enough for you to just wheel it off and diassemble it but i dont remeber i being

jamieisboss, 2016-03-18:
Hope my calculations and googleings were of some use not just inane babbling, 

I assume from the photos it went alright, can i ask what the weight was like in the end and how you managed it? As mentioned above, i have a similar scaffold project coming up so any experiential wisdom shared would be much appreciated

amcewen, 2016-03-21:
@jamieisboss In the end we hung it from their lighting rigging rather than the scaffolding poles, and it also rested on the floor as we decided that was safer than hanging all that weight from the frame of the board.

amcewen, 2016-03-21:
Any further items we need to buy are going to be best raised as individual issues as and when we encounter them now.

closed 2016-03-21 by amcewen