DoESLiverpool/Tosca #30: Generate SVG for the main performance pictograms

amcewen, 2016-03-01:
For the main performance we're drawing a map (see issue #20) and then overlaying that with pictograms that tie into specific locations and parts of the performance.  So we need to design(/source) SVG files for each of those.

I'll update this issue as we get the list from Lizzie, but so far... (check them off when done, and leave a comment with the location of the checked in file):

 * [ ] a mouse. It only needs to be an outline but something realistic rather than cartoonish

goatchurchprime, 2016-03-07:
We need these really soon so we can calibrate the time to draw them and their sizes.  

Get any examples now which look kind of artistic as a stand in so we're not always drawing "Hello world"?

Arty examples to chase up that would look appropriate someone there could chase up:

1) One of Ed's astrolabe patterns (possibly simplified so it draws in finite time)
2) Ed's 3 piece suit cloth cutting pattern, or a dress pattern.
3) A DXF technical drawing of some of the component designs for the motor boxes, or JD's monitor case
4) A nice circuit diagram from kicad of the schematic or the tracks

amcewen, 2016-03-09:
I've pinged Lizzie to see how things are going and to get some more things-to-draw from her, but @epsaul will hopefully be able to provide an astrolabe pattern or similar.  There's still the map of Liverpool city centre (I've not generated one for Toxteth yet) to try out - that's checked into this repo in the Examples folder.

amcewen, 2016-03-13:
We've got the full list now.  Any help in sourcing these much appreciated - either drop SVG files into the Examples folder in the code section, provide a link in a comment, or email me a file :-)

And tick off any you source, so everyone can see what still needs finding.

jackie1050, 2016-03-15:
Will update at lunchtime

jackie1050, 2016-03-17:
Only pigeon, cloud, fish, and cat have been approved by Lizzie so far. I've sent the others to her to see if they're what she's after

jackie1050, 2016-03-17:
All now approved - need to test esp. for time taken and size

jackie1050, 2016-03-17:
Final list of pictures and locations

Child
Fish on Spekeland Grove
Tree on Fern Grove
Mouse on Mozart Street

Lover
Pigeon on the junction of Rodney Street and Mount Street
Cat on Rice Street outside the Cracke
Balloon on Percy Street

Death
Bat on the junction of Upper Duke Street and Hope Street
Cloud on Stanhope Street
Dog on Hope Street

amcewen, 2016-03-21:
Done!  And performed.

closed 2016-03-21 by amcewen