DoESLiverpool/Tosca #31: Try out high intensity IR to communicate with the gondola instead of RF

goatchurchprime, 2016-03-01:
I'm getting a lot of unreliability with the jeenode RF communications now, once we took apart the old gondola.  Maybe the huge metal spanner in the works had been helping with reception.  

Anyway, we can bet on this RF completely failing at the critical moment, and none of us will have enough clues about how to fix it.  

I'd have more confidence with the IR comms tech outlined here: http://www.vishay.com/docs/80073/general.pdf
"""If we assume that the whole inner surface of a room (e.g. floor area of 30 m, height of 2.5 m) is irradiated with the emission of an infrared source with an overall irradiance of  e = 0.3 mW/m, then an emitted radiant flux of 35 mW is necessary (surface = 115 m, 100 % efficiency). With an 80 % reflection loss, about 175 mW of emitted radiation will be required for reliable reception in the whole room.

175 mW is a value, which can be achieved with an emitter TSAL6400 operating at a peak forward current of 500 mA. Under these conditions, no direct path between the emitter and the receiver is assumed, but that radiation will reach the detector after at least one reflection. This kind of remote control system is very user friendly for the customer because he can aim the handset in any direction of his living room. An IR emitter with a wide emitting angle will also provide this kind of comfortable remote control system"""

I feel that if we have high power narrower transmitting LEDs directed in range of the gondola whose IR detector is pointing in the right direction, then we will always be able to make it work.  

There's some highly developed protocols here, which involve intense high frequency bursts on the IR LEDs to get more signal brightness out of them and increase their range.  
    https://learn.sparkfun.com/tutorials/ir-communication

We should order the necessary items and program some 5V arduinos with enough power run stronger emitters and more sensitive detectors.  

The comms need only be one-way, which definitely simplifies things.

DefProc, 2016-03-01:
Just a thought, but are you setting the right frequency to match the radio
modules? They are tuned to a single frequency by one of the external
components on the module.

While they will work at other frequencies, the range is greatly reduced.

amcewen, 2016-03-03:
As both this and issue #24 were suggestions for things to try to improve problems with the z-axis movement, I've edited issue #24 to describe the problem rather than a possible solution, and going to mark this one as a duplicate.

Future comments to go over there...

closed 2016-03-03 by amcewen