DoESLiverpool/Tosca #39: Investigate some non-contact drawing technology

goatchurchprime, 2016-03-19:
Pen on paper without any force applied seems to be an issue.

Apparently you can get very tiny lasers that can engrave.  
   http://www.instructables.com/id/Pocket-laser-engraver/
We like lasers and they can be good for light night.  But they might be too slow (though worth a look at)

Alternatively there is paint spraying technology, which looks extremely good value for money:
    https://www.machinemart.co.uk/p/diy-air-brush-kit-cab1h/
    https://www.machinemart.co.uk/c/spray-guns-spraying-equipment-air-brushes/
You can get retractable hoses for the air supply, providing some pulldown
    https://www.machinemart.co.uk/p/clarke-9-metre-retractable-air-hose-reel-car/
(I can't find links to the air compressor product)

Can we find an artist who uses this tech already?  Paint stuff would be a mystery.  There might be some combination of paint substance and small laser curing to get the finer details.

jackie1050, 2016-04-25:
Airbrush, ink and propellant spray in my workshop box (should be moved to Tosca Polargraph box). Airbrush on indefinite loan so needs to be returned in same condition eventually. There seem to be cheap airbrush and compressor kits on eBay if this technology seems useful.

jackie1050, 2016-04-30:
Tried out airbrush on Thursday night - looks as though it will be good for writing as well as painting - need to look at line width etc. It worked well with a garden spray - Chris is going to bring in a fish tank pump to experiment with.

jamieisboss, 2016-05-13:
# **`etch-a-motherflipping-sketch`**

~~no clue how,~~ 

actually, i think i do, the gondola has a strong magnet , then corrugated plastic, I need to come back to this...