DoESLiverpool/Tosca #41: Investigate portable drawing surface for use at performances away from DoES

jackie1050, 2016-04-30:
Needs to be large (around 2m x 2m), stable, have a smooth surface and allow attachment of polargraph mechanism and paper roll mechanism if used.

A folding exhibition stand (e.g. http://www.ebay.co.uk/itm/Fordable-Banner-Pop-Roll-Pull-up-Sign-Exhibition-Display-Stand-Show-W-PVC-Screen-/281317070231?hash=item417fcdd597:g:qg4AAMXQyY1TVjNq) could be part of a potential solution, but needs further investigation.

jamieisboss, 2016-05-23:
As mentioned in #49 , I reckon a photographic backdrop holder with paper spool would be a good idea, remove the paper spool and add a metal bar and possibly an additional side piece for the middle and then aluminium foam-board seems to be all the rage for backgrounds at the moment, I've used it and it is pretty brill, lightweight but strong, you can print onto it, or paint/draw. 

You can also get it in bare metal if you ever wanted to try electrocuting it to see if that left cool markings and wasn't horribly dangerous in an uncontrollable way (if you earthed the panel so the foam didn't come into play so much you might be able to draw with a Tesla coil, and change the tone(?) so it sang or sounded like  weird speech. I've got no experience or basis for thinking this beyond you tube and an A Level in physics that I got a D in, so not qualified to tell you to shoot electricity at things and bear no responsibility if anyone tries it without asking someone more knowledgeable first,

Just realised you could just use aluminium sheet  for the zapping, and the foam one for regular stuff.

jackie1050, 2016-05-29:
Ordered banner as detailed above - should be delivered 3 - 8 June (photographic backdrop holders look good but more expensive, so try this first)

jamieisboss, 2016-05-30:
If you end up wanting the backdrop holder, i had another idea that is probably a fair bit cheaper that would work similarly too so let me know if you end up wanting something else and i'll cost it up and show you . 

What are you actually going to draw on with the display banner thing? Did you just get big paper and put it over the boards last time? 

I'm interested and i might see if we can borrow/rent the Tosca from you for the freshers fair in september, they didn't approve all my grant to build what I wanted but renting something big and interactive for the two days should seem more financially sensible to them. Are you planniing on having a free control mode, like a joystick or something? I should probably read all the docs and todos!
ruh the day

jackie1050, 2016-06-21:
The folding exhibition screen works fine

closed 2016-06-21 by jackie1050