DoESLiverpool/Tosca #43: Investigate real-time controllers

jackie1050, 2016-04-30:
Look into controllers such as etch-a-sketch, typewriter or scribble pad that could allow people to send instructions straight to the polargraph without programming

jamieisboss, 2016-05-30:
i had a thought that this is kind of like a flight control system for a quadcopter, maybe there's something in that as the new controllers have a lot of intelligence and here are a few open source ones, i'l see if i can find anything later on

Thingomy, 2016-05-30:
I have some (theoretical) experience with generating gcode on an arduino -- it would be a simple matter to have a bunch of input devices (joystics, rotery encoders etc) connected to an arduino, and it output gcode to tosca. What kind of controller is tosca running? is it grbl or what?