DoESLiverpool/Tosca #49: Design a motorized dispenser for paper roll

jackie1050, 2016-04-30:
This could be attached to the top of the board

jamieisboss, 2016-05-22:
if you used [photographers background paper](http://www.wexphotographic.com/buy-tetenal-super-white-background-paper-roll-2-72-x-11m/p1538567?mkwid=svdycxinl_dc&pcrid=89741321459&kword=&match=&plid=&gclid=cjp9kobg7swcfc2oaaodtcojha) it;s usualy heavy enough that you have to clamp it to stop  it unravelling,a really simple tapered plug or sprung out handles (like pogo pins) in one end should stop it from doing that if it's in rigght, then a cog and a gearmotor on the top pole he paper is sitting on should be simple enough.

That paper is pretty expensive, but it's huge and comes in hundreds of colours ad it's a preexisting use case there are [already systems you can buy](http://www.wexphotographic.com/buy-manfrotto-mn046mc-red-expan-with-red-metal-chain/p1011950) to control the roll(s) , i bet if i looked through that list there would be a remote control motorised one, Itd just be expensive and we can make one out of a bit of a bike and a pole.so bettter not to spend but nice to know [there are emergency options you could rent ](http://calumetrental.co.uk/background-support-kit)if something went pear shaped

jamieisboss, 2016-06-04:
I'll have a go at this if you want, do you have a link to the information about the paper roll you want to use? I'll work to that for weight and size.