DoESLiverpool/Tosca #50: Investigate software for producing images in format polargraph can use

jackie1050, 2016-04-30:
e.g. Processing has libraries to render an image as dots of varying size and/or colour

jackie1050, 2016-10-18:
StippleGen (http://wiki.evilmadscientist.com/StippleGen, https://github.com/evil-mad/stipplegen ) is open source software which uses Voronoi diagrams to generate TSP paths and SVGs of dot images (Darren has been using it to produce laser cut stencils for clay tiles).  
TSP paths could be very useful for producing images when Tosca's z-axis isn't working.
The source code is available on GitHub (Processing), so it might be possible to adapt it to calculate Voronoi regions with different shade numbers and use them for painting-by-numbers.

jamieisboss, 2017-07-16:
https://github.com/euphy/polargraphcontroller

jamieisboss, 2017-07-16:
https://github.com/euphy/polargraph_server_polarshield

jamieisboss, 2017-07-16:
https://github.com/cnc-club/gcodetools/blob/master/README.md

jamieisboss, 2017-07-16:
https://www.norwegiancreations.com/2015/08/an-intro-to-g-code-and-how-to-generate-it-using-inkscape/