DoESLiverpool/Tosca #53: Write instructions for new H-bridge direct voltage control system

goatchurchprime, 2016-10-07:
Servo motor drivers stripped out and Quadrature encoders wired directly into beaglebone, which has a new circuit board on it.  

Machinekit and G-codes no longer the way forward.

This enables experimental feedback loops to be made, and a direct web interface systems.

goatchurchprime, 2016-10-07:
First you need to log on to the tosca machine.  This requires the IPnumber.

Make sure the ehternet cable is properly in and the green led is flickering

Look for an open SSH port by doing:
    nmap -p22 192.168.0.255/24
and try logging on to several until you get to machinekit

Alternatively, using scanning for the MAC address  84:eb:18:e3:61:f4
sudo arp-scan 192.168.0.0/24 | grep 84:eb

Log on using:

ssh machinekit@192.168.0.67
password: machinekit

goatchurchprime, 2016-10-07:
The IDE we are developing in can be reached at:
    http://192.168.0.67:3000/ide.html

The code we have is in /var/lib/cloud9

First thing you need to do is enable the 2 equp readers (must be su):

echo bone_eqep0 > /sys/devices/bone_capemgr*/slots
echo bone_eqep1 > /sys/devices/bone_capemgr*/slots

Then you can run the script that serves the position and controls the motors

python polar1.py

serving on ...

Now you can go to the webpage that connects to this script and controls the motors

http://192.168.0.67/static/polarplot.html

This file is in  /var/lib/cloud9/static (can be edited through the IDE)

goatchurchprime, 2016-10-07:
Code for the polar plot (server and webpage) is herehttps://github.com/DoESLiverpool/Tosca/tree/master/beaglehbridge

goatchurchprime, 2016-10-25:
Current instructions:

1) Turn on the left switch at the mains (the BB only)
http://192.168.0.67:3000/ide.html

2) Find the ip-number:
sudo arp-scan 192.168.0.0/24 | grep 84:eb
ssh machinekit@192.168.0.67
password: machinekit

3) Go to http://192.168.0.45:3000/ide.html 
and using the shell command line window go to 
    /var/lib/cloud9/bbhquad/polarcode

and run:
    /usr/local/opt/python-3.4.5/bin/python3.4 mainpolar.py

4) go to http://192.168.0.45/static/polarmotion.html
With the websocket connected, see how you can drag the yellow disk and have the red disk follow it

DefProc, 2016-11-07:
@goatchurchprime Seeing as you're talking feedback loops on stepper motors, have you seen: [Mechaduino](https://hackaday.io/project/11224-mechaduino) yet?